nopnop2002/STM32_TFT_8bit #4: My TFT LCD 2.8" ili9325 don't work, only white screen

laserayaneh, 2018-05-20:
Hi,
I have a TFT LCD 2.8", this TFTLCD work fine with Arduino Uno and Arduino Pro Mini and other Arduino board with library and examples of https://github.com/adafruit/TFTLCD-Library
Wiring in Arduino boards is:
LCD_CS --> A3, LCD_CD --> A2, LCD_WR --> A1, LCD_RD --> A0, LCD_RESET --> A4
and 8 data lines is:
D0 --> 8,  D1 --> 9, D2 --> 2, D3 --> 3, D4 --> 4, D5 --> 5, D6 --> 6, D7 -->7

Now I have an STM32 (STM32F103C8T6) header board and wiring is:

LCD_RST ---> PB7, LCD_CS ---> PB6, LCD_RS ---> PB5, LCD_WR ---> PB4, LCD_RD ---> PB3
LCD_D0 ---> PA0, LCD_D1 ---> PA1, LCD_D2 ---> PA2, LCD_D3 ---> PA3, LCD_D4 ---> PA4, LCD_D5 ---> PA5, LCD_D6 ---> PA6, LCD_D7 ---> PA7, 5V ---> 5V, 3.3V ---> 3.3V, GND ---> GND

what is the problem, why my TFTLCD is only white ?!

I use this code:
https://github.com/nopnop2002/STM32_TFT_8bit/blob/master/examples/GraphicsTest_240x320_SMT32/GraphicsTest_240x320_SMT32.ino

Thanks so much

nopnop2002, 2018-06-02:
Please try the following sketch using UNO.

https://github.com/prenticedavid/MCUFRIEND_kbv/blob/master/examples/LCD_ID_readreg/LCD_ID_readreg.ino

Please tell me the result.

laserayaneh, 2018-06-04:
Thanks so much for your replay.

this TFTLCD ili9325 work fine with AVR series, but I want work this TFTLCD with STM32 series.

almost I will test this source with Arduino Uno and STM32 header...
I hope I can lunch with STM32...

Best wishes

nopnop2002, 2018-06-05:
There is a possibility that TFT you have isn't ILI9341.

So please change https://github.com/adafruit/TFTLCD-Library/blob/master/examples/graphicstest/graphicstest.pde

  tft.reset();
  uint16_t identifier = tft.readID();
  **Serial.println("identifier=" + String(identifier)); // ADD THIS LINE**

And Execute this with UNO.

laserayaneh, 2018-06-07:
Hi, 
This is output of this link:
https://github.com/prenticedavid/MCUFRIEND_kbv/blob/master/examples/LCD_ID_readreg/LCD_ID_readreg.ino

Thanks so much

Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) C4 C4 5A A6 00	Status Register
reg(0x000A) D4 D4	Get Power Mode
reg(0x000C) 55 55	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 28 94 03 08 08 10 00 15 48 00 07 00	Adjust Control 2
reg(0x00F6) 80 00 00 00	Interface Control

nopnop2002, 2018-06-07:
You'r TFT  isn't ILI9341.

reg(0x00D3) 00 00 00 00	ILI9341, ILI9488

True 9341 will be set this register.

laserayaneh, 2018-06-08:
Hi,
I use this code: 

`// Start Code
// ILI9341 ID Checker against data sheet http://www.adafruit.com/datasheets/ILI9341.pdf
// Ugly bit basher - R Jennings 11/11/14 (http://www.shaders.co.uk)
//
 #define CS PB6   // Chip Select goes to Analog 3
 #define CD PB5   // Command/Data goes to Analog 2
 #define WR PB4   // LCD Write goes to Analog 1
 #define RD PB3   // LCD Read goes to Analog 0
 #define RST PB7  // Can alternately just connect to Arduino's reset pin
 #define D0 PA0    // PA0
 #define D1 PA1    // PA1
 #define D2 PA2    // PA2
 #define D3 PA3    // PA3
 #define D4 PA4    // PA4
 #define D5 PA5    // PA5
 #define D6 PA6    // PA6
 #define D7 PA7    // PA7

void setup(void) {
// Set up port directions and default states
  pinMode(CS, OUTPUT);  pinMode(CD, OUTPUT);  pinMode(WR, OUTPUT);  pinMode(RD, OUTPUT);  pinMode(RST, OUTPUT);
  digitalWrite(CS, 1); digitalWrite(CD, 1); digitalWrite(WR, 1); digitalWrite(RD, 1); digitalWrite(RST, 1);
  pinMode(D0, OUTPUT); pinMode(D1, OUTPUT); pinMode(D2, OUTPUT); pinMode(D3, OUTPUT);
  pinMode(D4, OUTPUT); pinMode(D5, OUTPUT); pinMode(D6, OUTPUT); pinMode(D7, OUTPUT);

// This just drivers the serial monitor no screen activity  
  Serial.begin(9600);
  Serial.println("TFT LCD test - Read of the ID register");
//HW Reset
  digitalWrite(RST,0); delay(5);
  digitalWrite(RST,1); delay(500);  
// Read the ID Register, D3h
// Start by writing to the command register with the register to read
  digitalWrite(CD, 0); 
  digitalWrite(CS, 0);
  digitalWrite(D0, 1); digitalWrite(D1, 1); digitalWrite(D2, 0); digitalWrite(D3, 0);  // 3 - Change for lower nibble as required 
  digitalWrite(D4, 1); digitalWrite(D5, 0); digitalWrite(D6, 1); digitalWrite(D7, 1);  // D - Change for upper nibble as required
  digitalWrite(WR, 0); digitalWrite(WR, 1);
  digitalWrite(CS, 1);    
// Switch to reading   
  pinMode(D0, INPUT);    pinMode(D1, INPUT);    pinMode(D2, INPUT);    pinMode(D3, INPUT);
  pinMode(D4, INPUT);    pinMode(D5, INPUT);    pinMode(D6, INPUT);    pinMode(D7, INPUT);
  int x=0;
// Then read the four bytes and print to the serial port
// See section 8.3.23 page 233 for read sequence
// 1st Parameter - Dummy byte could be anything
  digitalWrite(CD, 1); 
  digitalWrite(CS, 0); 
  digitalWrite(RD, 0); 
  x=0;
  digitalWrite(RD, 0); 
  x|=digitalRead(D0);    x|=digitalRead(D1)<<1; x|=digitalRead(D2)<<2; x|=digitalRead(D3)<<3;
  x|=digitalRead(D4)<<4; x|=digitalRead(D5)<<5; x|=digitalRead(D6)<<6; x|=digitalRead(D7)<<7;
  Serial.print("Dummny Byte Read =        "); Serial.println(x, HEX);   
  digitalWrite(RD, 1); 
 // 2nd Parameter should be 00h
  x=0;
  digitalWrite(RD, 0); 
  x|=digitalRead(D0);    x|=digitalRead(D1)<<1; x|=digitalRead(D2)<<2; x|=digitalRead(D3)<<3;
  x|=digitalRead(D4)<<4; x|=digitalRead(D5)<<5; x|=digitalRead(D6)<<6; x|=digitalRead(D7)<<7;
  Serial.print("2nd Parameter Byte Read = "); Serial.println(x, HEX);   
  digitalWrite(RD, 1); 
 // 3rd Parameter should be 93h    
  x=0;
  digitalWrite(RD, 0); 
  x|=digitalRead(D0);    x|=digitalRead(D1)<<1;  x|=digitalRead(D2)<<2; x|=digitalRead(D3)<<3;
  x|=digitalRead(D4)<<4; x|=digitalRead(D5)<<5;  x|=digitalRead(D6)<<6; x|=digitalRead(D7)<<7;
  Serial.print("3rd Parameter Byte Read = "); Serial.println(x, HEX);   
  digitalWrite(RD, 1); 
 // 4th Parameter should be 41h     
  x=0;
  digitalWrite(RD, 0); 
  x|=digitalRead(D0);    x|=digitalRead(D1)<<1; x|=digitalRead(D2)<<2; x|=digitalRead(D3)<<3;
  x|=digitalRead(D4)<<4; x|=digitalRead(D5)<<5; x|=digitalRead(D6)<<6; x|=digitalRead(D7)<<7;
  Serial.print("4th Parameter Byte Read = "); Serial.println(x, HEX);   
  digitalWrite(RD, 1);  
}
//Nothing to do here
void loop(void) {
}
// End Code
`
and show this results:
`
TFT LCD test - Read of the ID register
Dummny Byte Read =        D3
2nd Parameter Byte Read = D3
3rd Parameter Byte Read = D3
4th Parameter Byte Read = D3
`

Thanks so much

laserayaneh, 2018-06-08:
Hi,
When I use this code: https://github.com/JoaoLopesF/SPFD5408/blob/master/examples/spfd5408_graphicstest/spfd5408_graphicstest.ino on my Arduino Uno, all thing is Okay and show "Found ILI9325 LCD driver" and its true.

Thanks so much

laserayaneh, 2018-06-09:
Thank you so much,

I am wait for your replay :-)

nopnop2002, 2018-06-09:
I tried your code.

My result:

Dummny Byte Read = 0
2nd Parameter Byte Read = 0
3rd Parameter Byte Read = 93
4th Parameter Byte Read = 41

Please see 8.3.23 chapter of this data seat.

https://cdn-shop.adafruit.com/datasheets/ILI9341.pdf

Your TFT is ILI9341, but a register read isn't made.

So please change your code.

https://github.com/nopnop2002/STM32_TFT_8bit/blob/master/examples/GraphicsTest_240x320_SMT32/GraphicsTest_240x320_SMT32.ino

void setup() {
  delay(1000);
  Serial.begin(9600);
  Serial.println("STM32_TFT_8bit Test!"); 

  ID = tft.readID();
  Serial.print("Device ID: 0x"); Serial.println(ID, HEX);
  **ID = 0x9341; // -----> ADD**
  tft.begin(ID);

laserayaneh, 2018-06-11:
Hi,

I set ID = 0x9341 and ID = x9325...

but my TFTLCD ili9325 not work !

my TFTLCD show this results:

**TFT LCD test - Read of the ID register
Dummny Byte Read = D3
2nd Parameter Byte Read = D3
3rd Parameter Byte Read = D3
4th Parameter Byte Read = D3**

Thanks so much

nopnop2002, 2018-06-11:
Is it work this??

https://github.com/adafruit/TFTLCD-Library/blob/master/examples/graphicstest/graphicstest.pde

Please tell me the contents shown to Arduino-IDE's Serial monitor..

closed 2020-07-12 by nopnop2002