nopnop2002/STM32_TFT_8bit #5: TFT flickers STM32F103C8T6 and spdf5408 TFT2,4Zoll

TB1405, 2018-07-15:
With arduino uno and MCU_friend_KBV the screen works. 
But with STM32 and your libary and example code the tft only flickers. 
And the hex code is 0x5454 instead of 0x5408 
Could you help me

nopnop2002, 2018-07-15:
I checked MCUFRIEND_kbv.cpp.
But it don't have 0x5454.

TB1405, 2018-07-15:
It isnt a 0x5454 Chip it is 5408 but the stm32 interprated it false i think the is a Problem with the parallel Port.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/nopnop2002/STM32_TFT_8bit/issues/5#issuecomment-405120392",
"url": "https://github.com/nopnop2002/STM32_TFT_8bit/issues/5#issuecomment-405120392",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [nopnop2002/STM32_TFT_8bit]  TFT flickers STM32F103C8T6 and spdf5408 TFT2,4Zoll  (#5)",
"sections": [
{
"text": "",
"activityTitle": "**nopnop2002**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@nopnop2002",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"nopnop2002/STM32_TFT_8bit\",\n\"issueId\": 5,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"nopnop2002/STM32_TFT_8bit\",\n\"issueId\": 5\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/nopnop2002/STM32_TFT_8bit/issues/5#issuecomment-405120392"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 356617936\n}"
}
],
"themeColor": "26292E"
}
]

nopnop2002, 2018-07-15:
SPFD5408 work fine by my environment.