nopnop2002/STM32_TFT_8bit #6: stm32f1 and spfd5408 dont work, only white screen

DanikVodopianov, 2019-02-08:
Hi. can you help me with your lib, i run your examples, but it doesnt work, only white screen, i checked pins, all connected right

nopnop2002, 2019-02-09:
It's to know the model number of the chip used for your TFT module that you should do first.

Arduino
https://github.com/prenticedavid/MCUFRIEND_kbv/blob/master/examples/LCD_ID_readreg/LCD_ID_readreg.ino

Raspberry Pi
https://github.com/nopnop2002/lcd_id

DanikVodopianov, 2019-02-11:
i had this output
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 E3 00 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 06	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 02	Inversion Control
reg(0x00B6) 00 02 82 27 04	Display Control
reg(0x00B7) 00 06	Entry Mode Set
reg(0x00BF) 00 02 02 02 02 02	ILI9481, HX8357-B
reg(0x00C0) 00 15 11 11 11 11 11 11 11	Panel Control
reg(0x00C8) 00 04 04 04 04 04 04 04 04 04 04 04 04	GAMMA
reg(0x00CC) 00 84	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 01 01	NVM Read
reg(0x00D3) 00 00 93 40	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 E3	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 86 10 09 16 08 3C 5A 4A 05 0D 09 1B 1F 0F	GAMMA-P
reg(0x00E1) 00 00 24 2A 00 0E 04 3D 64 4E 04 0B 0A B1 37 0F	GAMMA-N
reg(0x00EF) 00 04 2A 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 81 00 00	Interface Control
so i have ili9327 driver?

but then i tried sketch from another issue and the output is 

TFT size is 240x320
identifier=49344
Unknown LCD driver chip: C0C0

nopnop2002, 2019-02-11:
This is real pfd5408.

reg(0x0000) 54 08	ID: ILI9320, ILI9325, ILI9335, ...

----------------------------------------------------------------

Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 54 08	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) C4 C4 C4 C4	Manufacturer ID
reg(0x0009) C9 C9 C9 C9 C9	Status Register
reg(0x000A) CA CA	Get Powsr Mode
reg(0x000C) CC CC	Get Pixel Format
reg(0x0061) E1 E1	RDID1 HX8347-G
reg(0x0062) E2 E2	RDID2 HX8347-G
reg(0x0063) E3 E3	RDID3 HX8347-G
reg(0x0064) E4 E4	RDID1 HX8347-A
reg(0x0065) E5 E5	RDID2 HX8347-A
reg(0x0066) E6 E6	RDID3 HX8347-A
reg(0x0067) E7 E7	RDID Himax HX8347-A
reg(0x0070) F0 F0	Panel Himax HX8347-A
reg(0x00A1) E1 E1 E1 E1 E1	RD_DDB SSD1963
reg(0x00B0) F0 F0	RGB Interface Signal Control
reg(0x00B4) F4 F4	Inversion Control
reg(0x00B6) F6 F6 F6 F6 F6	Display Control
reg(0x00B7) F7 F7	Entry Mode Set
reg(0x00BF) FF FF FF FF FF FF	ILI9481, HX8357-B
reg(0x00C0) C0 C0 C0 C0 C0 C0 C0 C0 C0	Panel Control
reg(0x00C8) C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8	GAMMA
reg(0x00CC) CC CC	Panel Control
reg(0x00D0) D0 D0 D0	Power Control
reg(0x00D2) D2 D2 D2 D2 D2	NVM Read
reg(0x00D3) D3 D3 D3 D3	ILI9341, ILI9488
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) DC DC	RDID3
reg(0x00E0) E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0	GAMMA-P
reg(0x00E1) E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1	GAMMA-N
reg(0x00EF) EF EF EF EF EF EF	ILI9327
reg(0x00F2) F2 F2 F2 F2 F2 F2 F2 F2 F2 F2 F2 F2	Adjust Control 2
reg(0x00F6) F6 F6 F6 F6	Interface Control

nopnop2002, 2019-02-11:
Your TFT may be ILI9340.

reg(0x00D3) 00 00 93 40	ILI9341, ILI9488

Please try ILI9341.