nopnop2002/STM32_TFT_8bit #13: How use on stm32f401ccu6

brightproject, 2023-06-04:
How to use the library with kernel [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32) with stm32f401ccu6?
I have white dipslay.
https://github.com/Bodmer/TFT_eSPI/issues/2682

nopnop2002, 2023-06-04:
for [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32)

https://github.com/nopnop2002/Arduino-STM32-8bitTFT

brightproject, 2023-06-05:
Could you please change pins on?
```
#define TFT_CS   PB8  // Chip select control pin
#define TFT_DC   PB7  // Data Command control pin
#define TFT_RST  PB9  // Reset pin

#define TFT_WR   PB6  // Write strobe control pin 
#define TFT_RD   PB0  // Read pin
```

nopnop2002, 2023-06-05:
Tested TFT

ILI9325 2.4inch 240x320 TFT-Shield
ILI9341 2.4inch 240x320 TFT-Shield
ILI9342 2.4inch 240x320 TFT-Shield
SPFD5408 2.4inch 240x320 TFT-Shield
R61505 2.4inch 240x320 TFT-Shield
ST7783 2.4inch 240x320 TFT-Shield
LGDP4532 2.4inch 240x320 TFT-Shield
R61509V 3.6inch 240x400 TFT-Shield
ST7793 3.6inch 240x400 TFT-Shield
ILI9481 3.5inch 320x480 TFT-Shield
ILI9486 3.5inch 320x480 TFT-Shield
RM68140 3.95inch 320x480 TFT-Shield
ST7796 3.95inch 320x480 TFT-Shield
OPEN-SMART ILI9225 TFT-Shield
OPEN-SMART ILI9327 TFT-Shield
OPEN-SMART ILI9340 TFT-Shield

brightproject, 2023-06-05:
> ili9342 supports it.

How set to ILIL9342?
Library do it automatically?

nopnop2002, 2023-06-05:
>Library do it automatically?

Yes.
The driver is automatically determined here.
It will works if you change here.

https://github.com/nopnop2002/STM32_TFT_8bit/blob/master/STM32_TFT_8bit.cpp#L1588

brightproject, 2023-06-05:
> he driver is automatically determined here.
I soldered the display like this:
```
// common "MCUfriend" shields
#define TFT_CS   PB8  // Chip select control pin
#define TFT_DC   PB7  // Data Command control pin
#define TFT_RST  PB9  // Reset pin

#define TFT_WR   PB6  // Write strobe control pin 
#define TFT_RD   PB0  // Read pin

#define TFT_D0   PA0  // 8 bit parallel bus to TFT
#define TFT_D1   PA1
#define TFT_D2   PA2
#define TFT_D3   PA3
#define TFT_D4   PA4
#define TFT_D5   PA5
#define TFT_D6   PA6
#define TFT_D7   PA7
```
Can I change in the library, or do I need to re-solder the pins?

nopnop2002, 2023-06-05:
>Can I change in the library, or do I need to re-solder the pins?

https://github.com/nopnop2002/STM32_TFT_8bit/blob/master/STM32_TFT_8bit.h#L62

#define TFT_DATA       GPIOA
//Port data |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 |
//Pin stm32 |PA7|PA6|PA5|PA4|PA3|PA2|PA1|PA0|

//#define TFT_DATA       GPIOC
//Port data |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 |
//Pin stm32 |PC7|PC6|PC5|PC4|PC3|PC2|PC1|PC0|

Anything else cannot be used.

brightproject, 2023-06-05:
> Anything else cannot be used.
And here are the other pins.
https://github.com/nopnop2002/Arduino-STM32-8bitTFT#wirering-for-8bit-parallel-tft

closed 2023-06-13 by nopnop2002